linux-msm/qdl #7: How to use it properly to download?

baobei920722, 2019-11-12:
I compiled the project and used it. But I don't know how to use it to download the MDM9607 device image. I tried to read the code, and since I didn't understand the qualcomm download protocol, I couldn't read the code.
I use the following command:
./qdl prog_nand_firehose_9x07.mbn rawprogram_nand_p2K_b128K.xml patch_p2K_b128K.xml --include=./ --debug
Terminal output:[PROGRAM] errors while parsing program
I didn't find it in the XML file:"file_sector_offset","label"......

Can you give me detailed instructions on how to use it?

C457, 2019-11-12:
> I compiled the project and used it. But I don't know how to use it to download the MDM9607 device image. I tried to read the code, and since I didn't understand the qualcomm download protocol, I [...]

Your xml must have the contain these lines or it will fail to parse. https://github.com/andersson/qdl/blob/master/patch.c#L70
I also recommend trying patch0 and rawprogram by themselves

baobei920722, 2019-11-14:
@C457 :
[20191114-01-ubun18.04.txt](https://github.com/andersson/qdl/files/3844382/20191114-01-ubun18.04.txt)
I modified the code:
patch.c   patch_load function：
`		patch->sector_size = attr_as_unsigned(node, "SECTOR_SIZE_IN_BYTES", &errors);
		patch->byte_offset = 0;//attr_as_unsigned(node, "byte_offset", &errors);///modify
		patch->filename = attr_as_string(node, "filename", &errors);
		patch->partition = attr_as_unsigned(node, "physical_partition_number", &errors);
		patch->size_in_bytes = 0;//attr_as_unsigned(node, "size_in_bytes", &errors);//modify
		patch->start_sector = attr_as_string(node, "start_sector", &errors);
		patch->value = attr_as_string(node, "value", &errors);
		patch->what = attr_as_string(node, "what", &errors);`
program.c   program_load function:
`		program->sector_size = attr_as_unsigned(node, "SECTOR_SIZE_IN_BYTES", &errors);
		program->file_offset = 0;//attr_as_unsigned(node, "file_sector_offset", &errors);////modify
		program->filename = attr_as_string(node, "filename", &errors);
		program->label = 0;//attr_as_string(node, "label", &errors);/////modify
		program->num_sectors = attr_as_unsigned(node, "num_partition_sectors", &errors);
		program->partition = attr_as_unsigned(node, "physical_partition_number", &errors);
		program->start_sector = attr_as_string(node, "start_sector", &errors);`
I use the following command:
./qdl prog_nand_firehose_9x07.mbn rawprogram_nand_p2K_b128K.xml patch_p2K_b128K.xml --include=./ --debug
Got the above log，Error info：
qdl: failed to read: Connection timed out
I don't know how to handle this error.
My test environment runs the VM on Windows 10 and installs Ubuntu 18.04 in the VM, which I tested on Ubuntu. I don't know if this error is related to my environment.

baobei920722, 2019-11-15:
I tested after installing Ubuntu 18.04 on a new computer;
Its error log is the same as running Ubuntu 18.04 on the VM.
I don't know how to handle this error now.

jwinarske, 2021-05-02:
@baobei920722 I suspect you need to disable ModemManager.  

`systemctl stop ModemManager`

Then power cycle the target to put EDL into a known state.

closed 2025-09-09 by andersson