linux-msm/qdl #13: What does it means: LOG: VIP - Validated Image Programming is enabled - Validation is enabled.

kwaiching, 2020-08-02:
`$ ./qdl --storage ufs prog_emmc_ufs_firehose_Sdm660_ddr.elf rawprogram0.xml patch0.xml --include=./ --debug
HELLO version: 0x2 compatible: 0x1 max_len: 1024 mode: 0
READ64 image: 13 offset: 0x0 length: 0x40
READ64 image: 13 offset: 0x40 length: 0x2d8
READ64 image: 13 offset: 0x1000 length: 0x1000
READ64 image: 13 offset: 0x2000 length: 0xac8
READ64 image: 13 offset: 0x3000 length: 0x1000
READ64 image: 13 offset: 0x4000 length: 0x1000
READ64 image: 13 offset: 0x5000 length: 0x1000
READ64 image: 13 offset: 0x6000 length: 0x1000
READ64 image: 13 offset: 0x7000 length: 0x1000
READ64 image: 13 offset: 0x8000 length: 0x1000
READ64 image: 13 offset: 0x9000 length: 0x1000
READ64 image: 13 offset: 0xa000 length: 0x1000
READ64 image: 13 offset: 0xb000 length: 0x1000
READ64 image: 13 offset: 0xc000 length: 0x1000
READ64 image: 13 offset: 0xd000 length: 0x1000
READ64 image: 13 offset: 0xe000 length: 0x1000
READ64 image: 13 offset: 0xf000 length: 0x1000
READ64 image: 13 offset: 0x10000 length: 0x1000
READ64 image: 13 offset: 0x11000 length: 0x1000
READ64 image: 13 offset: 0x12000 length: 0x1000
READ64 image: 13 offset: 0x13000 length: 0x1000
READ64 image: 13 offset: 0x14000 length: 0x1000
READ64 image: 13 offset: 0x15000 length: 0x1000
READ64 image: 13 offset: 0x16000 length: 0x1000
READ64 image: 13 offset: 0x17000 length: 0x1000
READ64 image: 13 offset: 0x18000 length: 0x1000
READ64 image: 13 offset: 0x19000 length: 0x1000
READ64 image: 13 offset: 0x1a000 length: 0x1000
READ64 image: 13 offset: 0x1b000 length: 0x1000
READ64 image: 13 offset: 0x1c000 length: 0x1000
READ64 image: 13 offset: 0x1d000 length: 0x1000
READ64 image: 13 offset: 0x1e000 length: 0x1000
READ64 image: 13 offset: 0x1f000 length: 0x1000
READ64 image: 13 offset: 0x20000 length: 0x1000
READ64 image: 13 offset: 0x21000 length: 0x1000
READ64 image: 13 offset: 0x22000 length: 0x1000
READ64 image: 13 offset: 0x23000 length: 0x1000
READ64 image: 13 offset: 0x24000 length: 0x1000
READ64 image: 13 offset: 0x25000 length: 0x1000
READ64 image: 13 offset: 0x26000 length: 0x1000
READ64 image: 13 offset: 0x27000 length: 0x1000
READ64 image: 13 offset: 0x28000 length: 0x1000
READ64 image: 13 offset: 0x29000 length: 0x1000
READ64 image: 13 offset: 0x2a000 length: 0x1000
READ64 image: 13 offset: 0x2b000 length: 0x1000
READ64 image: 13 offset: 0x2c000 length: 0x1000
READ64 image: 13 offset: 0x2d000 length: 0x1000
READ64 image: 13 offset: 0x2e000 length: 0x1000
READ64 image: 13 offset: 0x2f000 length: 0x1000
READ64 image: 13 offset: 0x30000 length: 0x1000
READ64 image: 13 offset: 0x31000 length: 0x1000
READ64 image: 13 offset: 0x32000 length: 0x1000
READ64 image: 13 offset: 0x33000 length: 0x1000
READ64 image: 13 offset: 0x34000 length: 0x1000
READ64 image: 13 offset: 0x35000 length: 0x1000
READ64 image: 13 offset: 0x36000 length: 0x1000
READ64 image: 13 offset: 0x37000 length: 0x1000
READ64 image: 13 offset: 0x38000 length: 0x1000
READ64 image: 13 offset: 0x39000 length: 0x1000
READ64 image: 13 offset: 0x3a000 length: 0x1000
READ64 image: 13 offset: 0x3b000 length: 0x1000
READ64 image: 13 offset: 0x3c000 length: 0x1000
READ64 image: 13 offset: 0x3d000 length: 0x1000
READ64 image: 13 offset: 0x3e000 length: 0x1000
READ64 image: 13 offset: 0x3f000 length: 0x1000
READ64 image: 13 offset: 0x40000 length: 0x1000
READ64 image: 13 offset: 0x41000 length: 0x1000
READ64 image: 13 offset: 0x42000 length: 0x1000
[... 177 lines cut ...]
LOG: Failed to authenticate Digital Signature, resetting validation state
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="ERROR 11: Line 813: DIGITAL_SIGNATURE_DID_NOT_PASS"/>
</data>
LOG: ERROR 11: Line 813: DIGITAL_SIGNATURE_DID_NOT_PASS
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<response value="NAK" />
</data>
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="Reset validation state. Expecting Digital signature"/>
</data>
LOG: Reset validation state. Expecting Digital signature
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<response value="NAK" />
</data>
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?>
<data>
<log value="VIP - Validated Image Programming is enabled - Validation is enabled."/>
</data>
LOG: VIP - Validated Image Programming is enabled - Validation is enabled.
`

22mahmoud, 2021-04-20:
did you solve your problem yey?

kwaiching, 2021-04-21:
> did you solve your problem yey?

not yet

jwinarske, 2021-04-27:
@kwaiching is this device fused?

closed 2025-09-09 by andersson