linux-msm/qdl #15: Waiting for EDL Device

luke-shields, 2020-12-17:
Hi there,
I am attempting to flash a Jiophone 2 F300B Device however the QDL tool doesn't seem to be recognising it

`./qdl --debug --storage emmc --include '~/Downloads/LYF-F300B-001-01-21-230818' '/home/luke/Downloads/LYF-F300B-001-01-21-230818/boot_images/build/ms/bin/8909/emmc/unsigned/prog_emmc_firehose_8909_ddr.mbn' '/home/luke/Downloads/LYF-F300B-001-01-21-230818/common/build/bin/asic/sparse_images/rawprogram_unsparse.xml' '/home/luke/Downloads/LYF-F300B-001-01-21-230818/common/build/patch0.xml' 
Waiting for EDL device
/dev/bus/usb/001/009
/dev/bus/usb/001/009
`

Despite the device reading the correct ID Vender and ID Product

`Bus 001 Device 009: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
`

jwinarske, 2021-04-27:
Is the device in question fused?

SnikhilK, 2023-06-09:
Faced a similar issue and this solved it~

Note the correct ID Vendor and Product ID from `lsusb`
Ex: Bus 001 Device 010: ID **05c6**:**9008** Qualcomm, Inc. Gobi Wireless Modem (QDL mode)

Modify the android.rules `sudo nano /etc/udev/rules.d/51-android.rules`

Add this line, if not added already:
SUBSYSTEM=="usb", ATTR{idVendor}=="05c6", ATTR{idProduct}=="**_XXXX_**", MODE="0666", GROUP="plugdev"

Here replace XXXX --> with the Product ID from lsusb output, ex:9008

Now reload the rules :
```
sudo udevadm control --reload-rules
sudo service udev restart
sudo udevadm trigger
```

jwinarske, 2023-06-09:
Also be sure to disable modem manager

jwinarske, 2023-11-28:
@Andoramb your issue isn't related to the topic.  Create a new issue please.  In your case I suspect your virtual environment is not correctly setup for USB.  You have a number of options to confirm that.  Also strace -f is your friend.  Also make sure modem manager is not running.  Create new issue with correct title, and put your results there.  Cheers

Andoramb, 2023-11-28:
Thanks @jwinarske!
I thought it might be related, but there are a lot of variables here.
I'll doublecheck and create a new issue, if it's still relevant.

Thanks for the quick response 👍

closed 2024-05-09 by andersson