linux-msm/qdl #19: APQ8016 EDL

jwinarske, 2021-04-22:
Attempting to use qdl (master) with APQ8016 - DB410c.

Device enumeration:
`Bus 007 Device 027: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)`

Using dragonboard-410c-bootloader-emmc-linux-159.zip I issue:
 `sudo /b/qdl/qdl prog_emmc_firehose_8916.mbn rawprogram0.xml patch0.xml`

It just returns fairly quick, and says nothing.

[strace log](https://gist.github.com/jwinarske/882b4971e1048bad6ab76df9a00a34d1)

log tail

    ioctl(4, USBDEVFS_IOCTL, 0x7ffc02ac36f0) = -1 ENODATA (No data available)
    ioctl(4, USBDEVFS_CLAIMINTERFACE, 0x7ffc02ac36cc) = 0
    openat(AT_FDCWD, "prog_emmc_firehose_8916.mbn", O_RDONLY) = 3
    ioctl(4, USBDEVFS_BULK, 0x7ffc02ac25d0) = -1 ETIMEDOUT (Connection timed out)

Does prog_emmc_firehose_8916.mbn require an update to work with master?

jwinarske, 2021-05-02:
The first `qdl_read` in `sahara_run` is timing out.  It's not get a response from device while in EDL mode, and timing out without error or warning.

jwinarske, 2021-05-02:
Same issue on Fedora 33 and Ubuntu 18.  I stopped ModemManager on each.  What am I missing? @ndechesne @lumag

jwinarske, 2021-05-02:
Well I didn't cycle the USB after turning off ModemManager, so SoC was confused.  After power cycling SoC it worked as expected on Ubuntu 18.

closed 2021-05-02 by jwinarske

jwinarske, 2021-05-02:
also confirmed to work on Fedora 33.  Test were all run on master as of today.

CyberTailor, 2023-06-14:
Getting "23: Connection timed out"