linux-msm/qdl #20: [db410c] 10 Second delay

jwinarske, 2021-05-05:
I'm finding some odd interactions with timing.  I suspect it's related to the programmer (target flash stub) being used.

Once the flash stub is pushed via sahara it jumps into it's entry point.  Depending on the flash stub used it may need to train the DDR, etc.  So completion time varies on flash stub implementation (target implementation).

The current timeout/delay:  https://github.com/andersson/qdl/blob/master/firehose.c#L617
I find dropping this timeout to 500 ms works reliably.  This delay corresponds to the first entry of the listed log.

Then a default configuration is sent with a big default value.  In the case of the apq8016 emmc flash stub it causes a timeout after re-config.  Seems it might be better to read then modify, which would eliminate this additional timeout/delay.

Delay is ~10 seconds

```
qdl: firehose operation timed out
FIREHOSE WRITE: <?xml version="1.0"?>
<data><configure MemoryName="emmc" MaxPayloadSizeToTargetInBytes="1048576" verbose="0" ZLPAwareHost="1" SkipStorageInit="0"/></data>

FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="Host's payload to target size is too large" /></data>
LOG: Host's payload to target size is too large
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDC0 fh@0x08019C28" /></data>
LOG: logbuf@0x0801CDC0 fh@0x08019C28
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><response value="NAK" MinVersionSupported="1" MemoryName="eMMC" MaxPayloadSizeFromTargetInBytes="4096" MaxPayloadSizeToTargetInBytes="16384" MaxPayloadSizeToTargetInBytesSupported="16384" MaxXMLSizeInBytes="4096" Version="1" TargetName="8916" /></data>
FIREHOSE WRITE: <?xml version="1.0"?>
<data><configure MemoryName="emmc" MaxPayloadSizeToTargetInBytes="16384" verbose="0" ZLPAwareHost="1" SkipStorageInit="0"/></data>

ERROR: n = -1, errno = 110 (Connection timed out)
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDC0 fh@0x08019C28" /></data>
LOG: logbuf@0x0801CDC0 fh@0x08019C28
qdl: firehose operation timed out
FIREHOSE WRITE: <?xml version="1.0"?>
<data><configure MemoryName="emmc" MaxPayloadSizeToTargetInBytes="16384" verbose="0" ZLPAwareHost="1" SkipStorageInit="0"/></data>

FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDC0 fh@0x08019C28" /></data>
LOG: logbuf@0x0801CDC0 fh@0x08019C28
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><response value="ACK" MinVersionSupported="1" MemoryName="eMMC" MaxPayloadSizeFromTargetInBytes="4096" MaxPayloadSizeToTargetInBytes="16384" MaxPayloadSizeToTargetInBytesSupported="16384" MaxXMLSizeInBytes="4096" Version="1" TargetName="8916" /></data>
[CONFIGURE] max payload size: 16384
FIREHOSE WRITE: <?xml version="1.0"?>
<data><program SECTOR_SIZE_IN_BYTES="512" num_partition_sectors="1" physical_partition_number="0" start_sector="131072" filename="sbc_1.0_8016.bin"/></data>

FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="start 131072, num 1" /></data>
LOG: start 131072, num 1
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><response value="ACK" rawmode="true" /></data>

```

moctaviano, 2023-11-20:
localhost:/home/moctaviano/Downloads/Lenovo_Yoga_Tab_3_10_YT3-X50FQDL/Firmware # qdl --debug --storage emmc --include '/home/moctaviano/Downloads/Lenovo_Yoga_Tab_3_10_YT3-X50FQDL/Firmaware/' '/home/moctaviano/Downloads/Lenovo_Yoga_Tab_3_10_YT3-X50FQDL/Firmware/prog_emmc_firehose_8909_ddr.mbn' '/home/moctaviano/Downloads/Lenovo_Yoga_Tab_3_10_YT3-X50FQDL/Firmware/rawprogram_unsparse.xml' '/home/moctaviano/Downloads/Lenovo_Yoga_Tab_3_10_YT3-X50FQDL/Firmware/patch0.xml'
HELLO version: 0x2 compatible: 0x1 max_len: 1024 mode: 0
READ image: 13 offset: 0x0 length: 0x34
READ image: 13 offset: 0x34 length: 0x120
READ image: 13 offset: 0x1000 length: 0x148
READ image: 13 offset: 0x418cc length: 0x1000
READ image: 13 offset: 0x428cc length: 0x1000
READ image: 13 offset: 0x438cc length: 0x998
READ image: 13 offset: 0x44264 length: 0x978
READ image: 13 offset: 0x2000 length: 0x1000
READ image: 13 offset: 0x3000 length: 0x1000
READ image: 13 offset: 0x4000 length: 0x1000
READ image: 13 offset: 0x5000 length: 0x1000
READ image: 13 offset: 0x6000 length: 0x1000
READ image: 13 offset: 0x7000 length: 0x1000
READ image: 13 offset: 0x8000 length: 0x1000
READ image: 13 offset: 0x9000 length: 0x1000
READ image: 13 offset: 0xa000 length: 0x1000
READ image: 13 offset: 0xb000 length: 0x1000
READ image: 13 offset: 0xc000 length: 0x1000
READ image: 13 offset: 0xd000 length: 0x1000
READ image: 13 offset: 0xe000 length: 0x1000
READ image: 13 offset: 0xf000 length: 0x1000
READ image: 13 offset: 0x10000 length: 0x1000
READ image: 13 offset: 0x11000 length: 0x1000
READ image: 13 offset: 0x12000 length: 0x1000
READ image: 13 offset: 0x13000 length: 0x1000
READ image: 13 offset: 0x14000 length: 0x1000
READ image: 13 offset: 0x15000 length: 0x1000
READ image: 13 offset: 0x16000 length: 0x1000
READ image: 13 offset: 0x17000 length: 0x1000
READ image: 13 offset: 0x18000 length: 0x1000
READ image: 13 offset: 0x19000 length: 0x1000
READ image: 13 offset: 0x1a000 length: 0x1000
READ image: 13 offset: 0x1b000 length: 0x1000
READ image: 13 offset: 0x1c000 length: 0x1000
READ image: 13 offset: 0x1d000 length: 0x1000
READ image: 13 offset: 0x1e000 length: 0x1000
READ image: 13 offset: 0x1f000 length: 0x1000
READ image: 13 offset: 0x20000 length: 0x1000
READ image: 13 offset: 0x21000 length: 0x1000
READ image: 13 offset: 0x22000 length: 0x1000
READ image: 13 offset: 0x23000 length: 0x1000
READ image: 13 offset: 0x24000 length: 0x1000
READ image: 13 offset: 0x25000 length: 0x1000
READ image: 13 offset: 0x26000 length: 0x1000
READ image: 13 offset: 0x27000 length: 0x1000
READ image: 13 offset: 0x28000 length: 0x1000
READ image: 13 offset: 0x29000 length: 0x1000
READ image: 13 offset: 0x2a000 length: 0x1000
READ image: 13 offset: 0x2b000 length: 0x1000
READ image: 13 offset: 0x2c000 length: 0x1000
READ image: 13 offset: 0x2d000 length: 0x1000
READ image: 13 offset: 0x2e000 length: 0x1000
READ image: 13 offset: 0x2f000 length: 0x1000
READ image: 13 offset: 0x30000 length: 0x1000
READ image: 13 offset: 0x31000 length: 0x1000
READ image: 13 offset: 0x32000 length: 0x1000
READ image: 13 offset: 0x33000 length: 0x1000
READ image: 13 offset: 0x34000 length: 0x1000
READ image: 13 offset: 0x35000 length: 0x1000
READ image: 13 offset: 0x36000 length: 0x1000
READ image: 13 offset: 0x37000 length: 0x1000
READ image: 13 offset: 0x38000 length: 0x1000
READ image: 13 offset: 0x39000 length: 0x1000
READ image: 13 offset: 0x3a000 length: 0x1000
READ image: 13 offset: 0x3b000 length: 0x1000
READ image: 13 offset: 0x3c000 length: 0xdc
READ image: 13 offset: 0x3c0dc length: 0x1000
READ image: 13 offset: 0x3d0dc length: 0x1000
READ image: 13 offset: 0x3e0dc length: 0x1000
READ image: 13 offset: 0x3f0dc length: 0x1000
READ image: 13 offset: 0x400dc length: 0x1000
READ image: 13 offset: 0x410dc length: 0x7f0
END OF IMAGE image: 13 status: 0
DONE status: 0
qdl: firehose operation timed out
FIREHOSE WRITE: <?xml version="1.0"?>
<data><configure MemoryName="emmc" MaxPayloadSizeToTargetInBytes="1048576" verbose="0" ZLPAwareHost="1" SkipStorageInit="0"/></data>

ERROR: n = -1, errno = 19 (No such device)

andersson, 2025-09-17:
The way this was done changed over the years, but was effectively a 5 second delay with the ability of draining <log/> entries.
The code has now been changed to actively attempt to initiate the communication with the programmer during this window, vastly reducing the wait time...

closed 2025-09-17 by andersson