linux-msm/qdl #27: About xml prase

stellalio, 2022-01-12:
Hi，andersson

When I use the tool to download for 9607 and 9655 device it occurs：[PROGRAM] errors while parsing program.
So I think some unnecessary field parsing can make selective judgment.For example, 
program->sector_size = attr_as_unsigned(node, "SECTOR_SIZE_IN_BYTES", &errors);
could replace with :
if (!strcmp((char *)node->properties->name, "SECTOR_SIZE_IN_BYTES")) {
                program->sector_size = attr_as_unsigned(node, "SECTOR_SIZE_IN_BYTES", &errors);
}
If the device xml file doesn't have SECTOR_SIZE_IN_BYTES，it will not prase the field SECTOR_SIZE_IN_BYTES；if the xml have，it will prase then.About this change,I have test and it can be compatible with different devices to parse xml.

andersson, 2024-06-10:
"start_sector", "file_sector_offset" and "num_partition_sectors" are expressed in units of "SECTOR_SIZE_IN_BYTES", so this would require that we define a default value that works in all cases where one isn't explicitly defined.

konradybcio, 2025-07-01:
This can't really be fixed, as the device doesn't provide this data itself (and it varies between storage media and units)

closed 2025-07-01 by konradybcio