linux-msm/qdl #35: how to reboot after qdl flash

onlinesen, 2022-12-01:
wouldu pls add reboot to normal/power on funtion?

guoshouwang, 2023-02-14:
It already has reboot function, but maybe it can't works.
Please try this patch, it seems work but I don't know why.

diff --git a/firehose.c b/firehose.c
index 319bac8..d9ad588 100644
--- a/firehose.c
+++ b/firehose.c
@@ -652,6 +652,7 @@ int firehose_run(struct qdl_device *qdl, const char *incdir, const char *storage
                firehose_set_bootable(qdl, bootable);
 
        firehose_reset(qdl);
+       firehose_reset(qdl);
 
        return 0;
 }

ishan-kpit, 2025-06-25:
The patch is there in the master branch.
Yet the reset doesn't work.
Device doesn't reset after image flashing.

konradybcio, 2025-12-30:
Could you retry on the current state of master?

closed 2026-02-02 by andersson