linux-msm/qdl #40: --version

marado, 2023-04-03:
It would be nice to have a `--version` on this tool, specially if it is one that uses uses the git hash when it is a development version, so we can easilly pinpoint exactly what is on the qdl binary being used. Bonus points if `--debug` by default starts by outputing the version, so qdl logs will have that info.

ljkeller, 2025-01-16:
Seeing as qdl flashes seem to occasionally fail based upon the version of QDL used, I seriously agree agree with this request. Would this be something any user could implement?

ndechesne, 2025-01-17:
Yes, it makes sense. And we don't want that to be a pain to maintain in the repo, so ideally we could create a version string with "git describe --dirty --always --tags". We will check with our internal tools team who builds QDL for Windows and Linux (on the Qualcomm Software Center) to make sure that they are indeed using a git workspace in their build env, so that this would work for them too. The implementation will then be straightforward

PR #90 (2025-01-21): RFC: implement version information
Closes #40
closed 2025-01-21 by andersson