linux-msm/qdl #54: QDL Flash in a VM?

Andoramb, 2024-02-16:
Hey,
I'm trying to get this worked in a Proxmox-VM setup.
I'm running a turnkeycore image, but the same happened on the latest ubuntu server.

`Linux core 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux`

It seems it is unable to communicate with the device -> `Connection timed out`

- I tried to increase the timeout in [sahara.c - qdl_read](https://github.com/linux-msm/qdl/blob/3b22df2bc7de02d867334af3a7aa8606db4f8cdd/sahara.c#L258) but it just hangs.
- I tried to blacklist qcserial kernel module on both Proxmox host and the VM.
- It works flashing from host directly (and other regular conditions)
- I have hotplug enabled on the VM, tried to directly assign the port and the Vendor/Device IDs.
- I tried giving full access by adding a udev rule
- `SUBSYSTEM=="usb", ATTRS{idVendor}=="XXXX", ATTRS{idProduct}=="XXXX", MODE="0666"`

Strace gave me this:
```
write(2, "Waiting for EDL device\n", 23Waiting for EDL device
) = 23
pselect6(4, [3], NULL, NULL, NULL, NULL) = 1 (in [3])
recvfrom(3, NULL, 0, MSG_PEEK|MSG_TRUNC, NULL, NULL) = 1224
recvmsg(3, {msg_name={sa_family=AF_NETLINK, nl_pid=280, nl_groups=0x000002}, msg_namelen=128 => 12, msg_iov=[{iov_base=[{prefix="libudev", magic=htonl(0xfeedcafe), header_size=40, properties_off=40, properties_len=1184, filter_subsystem_hash=htonl(0x577c5e5), filter_devtype_hash=htonl(0x27f8f50c), filter_tag_bloom_hi=htonl(0x2080000), filter_tag_bloom_lo=htonl(0x400001)}, "UDEV_DATABASE_VERSION=1\0ACTION=a"...], iov_len=1224}], msg_iovlen=1, msg_control=[{cmsg_len=28, cmsg_level=SOL_SOCKET, cmsg_type=SCM_CREDENTIALS, cmsg_data={pid=2521, uid=0, gid=0}}], msg_controllen=32, msg_flags=0}, 0) = 1224
getrandom("\x28\x5d\xa9\xdb\x16\x26\x4c\x56\x4a\xd4\x66\x5e\x19\x0d\xe9\x60", 16, GRND_INSECURE) = 16
getrandom("\x55\xd4\x33\x5b\x53\x69\x1d\x70\xee\xe5\x9f\x47\xcf\xc7\x57\x69", 16, GRND_INSECURE) = 16
getrandom("\x66\x14\x81\x2f\x6a\xad\x79\x61\x40\xa6\x0c\xff\x87\x93\xf8\x20", 16, GRND_INSECURE) = 16
getrandom("\x4a\x8f\x0e\xdb\x5b\x42\xd0\xcf\x0c\x18\xbb\xab\xb7\x17\x90\x06", 16, GRND_INSECURE) = 16
openat(AT_FDCWD, "/dev/bus/usb/002/005", O_RDWR) = 4
read(4, "\22\1\0\2\0\0\0@\306\5\10\220\0\0\1\2\0\1\t\2 \0\1\1\0\200\1\t\4\0\0\2"..., 1024) = 50
close(3)                                = 0
ioctl(4, USBDEVFS_IOCTL, 0x7ffd7df19830) = -1 ENODATA (No data available)
ioctl(4, USBDEVFS_CLAIMINTERFACE, 0x7ffd7df1980c) = 0
openat(AT_FDCWD, "prog_firehose_ddr.elf", O_RDONLY) = 3
ioctl(4, USBDEVFS_BULK, 0x7ffd7df18710) = -1 ETIMEDOUT (Connection timed out)
close(3)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

Do you have any suggestions on how to investigate further? 😄 
Thanks!

Andoramb, 2024-09-25:
Oh, I stumbled on my own issue, while I was digging deeper for this 🙂

quic-kdybcio, 2025-07-01:
qdl uses libusb, so it's not really doing anything special - it may be due to your particular VM setup. Are you seeing any similar issues with other USB devices? Does a plugin-then-start-the-VM sequence result in any different behavior?

quic-kdybcio, 2025-07-01:
Well in any case, it uses libusb *now*, please try with a fresher release :)

Andoramb, 2025-07-08:
Hey @quic-kdybcio, thanks for the reply!
I don't remember exactly how I've solved it, but the root cause lies within the use of `libusb` and the way it's handling the devices -> there is a sudden reconnect of the devices and Proxmox misses to reattach the device to the VM properly.
In case of adb, a workaround was to skip `libusb` libs at all with `ADB_LIBUSB=0`.
With `qdl`, as this is not possible, the solution was to attach the whole PCI device with the USB ports on it and run the VM in privileged mode.

closed 2025-07-08 by Andoramb