linux-msm/qdl #76: qdl: bulk write failed: Operation timed out

RobusTetus, 2024-07-05:
Hello, I am having some difficulties with flashing a qualcomm device with this utility. After running `sudo qdl --debug prog_emmc_firehose_8953_ddr.mbn rawprogram0.xml patch0.xml` and connecting device in EDL mode produces following output. Any ideas on what could be the cause?

```Waiting for EDL device
USB: using out-chunk-size of 1048576
HELLO version: 0x2 compatible: 0x1 max_len: 1024 mode: 0
READ image: 13 offset: 0x0 length: 0x34
READ image: 13 offset: 0x34 length: 0x120
READ image: 13 offset: 0x1000 length: 0x1000
READ image: 13 offset: 0x2000 length: 0xa48
READ image: 13 offset: 0x59be0 length: 0x1000
READ image: 13 offset: 0x5abe0 length: 0xf10
READ image: 13 offset: 0x5baf0 length: 0x1000
READ image: 13 offset: 0x5caf0 length: 0x7f0
READ image: 13 offset: 0x3000 length: 0x1000
READ image: 13 offset: 0x4000 length: 0x1000
READ image: 13 offset: 0x5000 length: 0x1000
READ image: 13 offset: 0x6000 length: 0x1000
READ image: 13 offset: 0x7000 length: 0x1000
READ image: 13 offset: 0x8000 length: 0x1000
READ image: 13 offset: 0x9000 length: 0x1000
READ image: 13 offset: 0xa000 length: 0x1000
READ image: 13 offset: 0xb000 length: 0x1000
READ image: 13 offset: 0xc000 length: 0x1000
READ image: 13 offset: 0xd000 length: 0x1000
READ image: 13 offset: 0xe000 length: 0x1000
READ image: 13 offset: 0xf000 length: 0x1000
READ image: 13 offset: 0x10000 length: 0x1000
READ image: 13 offset: 0x11000 length: 0x1000
READ image: 13 offset: 0x12000 length: 0x1000
READ image: 13 offset: 0x13000 length: 0x1000
READ image: 13 offset: 0x14000 length: 0x1000
READ image: 13 offset: 0x15000 length: 0x1000
READ image: 13 offset: 0x16000 length: 0x1000
READ image: 13 offset: 0x17000 length: 0x1000
READ image: 13 offset: 0x18000 length: 0x1000
READ image: 13 offset: 0x19000 length: 0x1000
READ image: 13 offset: 0x1a000 length: 0x1000
READ image: 13 offset: 0x1b000 length: 0x1000
READ image: 13 offset: 0x1c000 length: 0x1000
READ image: 13 offset: 0x1d000 length: 0x1000
READ image: 13 offset: 0x1e000 length: 0x1000
READ image: 13 offset: 0x1f000 length: 0x1000
READ image: 13 offset: 0x20000 length: 0x1000
READ image: 13 offset: 0x21000 length: 0x1000
READ image: 13 offset: 0x22000 length: 0x1000
READ image: 13 offset: 0x23000 length: 0x1000
READ image: 13 offset: 0x24000 length: 0x1000
READ image: 13 offset: 0x25000 length: 0x1000
READ image: 13 offset: 0x26000 length: 0x1000
READ image: 13 offset: 0x27000 length: 0x1000
READ image: 13 offset: 0x28000 length: 0x1000
READ image: 13 offset: 0x29000 length: 0x1000
READ image: 13 offset: 0x2a000 length: 0x1000
READ image: 13 offset: 0x2b000 length: 0x1000
READ image: 13 offset: 0x2c000 length: 0x1000
READ image: 13 offset: 0x2d000 length: 0x1000
READ image: 13 offset: 0x2e000 length: 0x1000
READ image: 13 offset: 0x2f000 length: 0x1000
READ image: 13 offset: 0x30000 length: 0x1000
READ image: 13 offset: 0x31000 length: 0x1000
READ image: 13 offset: 0x32000 length: 0x1000
READ image: 13 offset: 0x33000 length: 0x1000
READ image: 13 offset: 0x34000 length: 0x1000
READ image: 13 offset: 0x35000 length: 0x1000
READ image: 13 offset: 0x36000 length: 0x1000
READ image: 13 offset: 0x37000 length: 0x1000
READ image: 13 offset: 0x38000 length: 0x1000
READ image: 13 offset: 0x39000 length: 0x1000
READ image: 13 offset: 0x3a000 length: 0x1000
READ image: 13 offset: 0x3b000 length: 0x1000
READ image: 13 offset: 0x3c000 length: 0x1000
READ image: 13 offset: 0x3d000 length: 0x1000
READ image: 13 offset: 0x3e000 length: 0x1000
READ image: 13 offset: 0x3f000 length: 0x1000
READ image: 13 offset: 0x40000 length: 0x1000
READ image: 13 offset: 0x41000 length: 0x1000
READ image: 13 offset: 0x42000 length: 0x1000
READ image: 13 offset: 0x43000 length: 0x1000
READ image: 13 offset: 0x44000 length: 0x1000
READ image: 13 offset: 0x45000 length: 0x1000
READ image: 13 offset: 0x46000 length: 0x1000
READ image: 13 offset: 0x47000 length: 0x1000
READ image: 13 offset: 0x48000 length: 0x1000
READ image: 13 offset: 0x49000 length: 0x1000
READ image: 13 offset: 0x4a000 length: 0x1000
READ image: 13 offset: 0x4b000 length: 0x1000
READ image: 13 offset: 0x4c000 length: 0x1000
READ image: 13 offset: 0x4d000 length: 0x1000
READ image: 13 offset: 0x4e000 length: 0x388
READ image: 13 offset: 0x4e388 length: 0x1000
READ image: 13 offset: 0x4f388 length: 0x1000
READ image: 13 offset: 0x50388 length: 0x1000
READ image: 13 offset: 0x51388 length: 0x1000
READ image: 13 offset: 0x52388 length: 0x1000
READ image: 13 offset: 0x53388 length: 0x1000
READ image: 13 offset: 0x54388 length: 0x1000
READ image: 13 offset: 0x55388 length: 0x1000
READ image: 13 offset: 0x56388 length: 0x1000
READ image: 13 offset: 0x57388 length: 0x1000
READ image: 13 offset: 0x58388 length: 0x1000
READ image: 13 offset: 0x59388 length: 0x858
END OF IMAGE image: 13 status: 0
DONE status: 0
FIREHOSE WRITE: <?xml version="1.0"?>
<data><configure MemoryName="ufs" MaxPayloadSizeToTargetInBytes="1048576" verbose="0" ZLPAwareHost="1" SkipStorageInit="0"/></data>

FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x08070078 fh@0x0806CEE0" /></data>
LOG: logbuf@0x08070078 fh@0x0806CEE0
FIREHOSE READ: <?xml version="1.0" encoding="UTF-8" ?><data><response value="ACK" MinVersionSupported="1" MemoryName="eMMC" MaxPayloadSizeFromTargetInBytes="4096" MaxPayloadSizeToTargetInBytes="1048576" MaxPayloadSizeToTargetInBytesSupported="1048576" MaxXMLSizeInBytes="4096" Version="1" TargetName="8953" /></data>
[CONFIGURE] max payload size: 1048576
FIREHOSE WRITE: <?xml version="1.0"?>
<data><program SECTOR_SIZE_IN_BYTES="512" num_partition_sectors="172032" physical_partition_number="0" start_sector="131072" filename="NON-HLOS.bin"/></data>

qdl: bulk write failed: Operation timed out
[PROGRAM] failed to write program command```

snubbaluba, 2024-10-25:
I'm having the same issue. Was there a solution for this?

taiten, 2024-12-04:
This happen to me, the problem may go away if switch to a native type-A USB port on the host PC.

Erikvv, 2024-12-20:
Maybe the flag `--out-chunk-size 4096` helps

quic-bjorande, 2025-09-17:
This looks identical to the problems reported/seen on MSM8916. Please test #139

andersson, 2025-10-24:
Depending on the sequence of things, the write might time out because there's unconsumed log entries, but this might also be a case of #145

closed 2025-11-11 by andersson