linux-msm/qdl #77: A problem about EMMC support

SlarkXiao, 2024-07-12:
When flash a partition with 'sparse="true"' flag,  no error returned.
But device can't boot up successfully and it reports corresponding partition mount failed.
According to the Windows tool, it has split image up into 4 partitions and then write 1 by 1.
It seems we don't obtain such mechanism.

Test log as below from windows QFIL:
xml:
<program start_sector="3031040" size_in_KB="262144.0" num_partition_sectors="524288"  filename="system.img" sparse="true" start_byte_hex="0x5c800000" label="system"/> 

QFIL log:
INFO: File system.img is a sparse file, being split up into 4 separate XML tags 
......
INFO: system.img is a sparse image   
INFO: Looking for file 'system.img'  
INFO: system.img is a sparse image   
INFO: Looking for file 'system.img'  
INFO: system.img is a sparse image   
INFO: Looking for file 'system.img'  
INFO: system.img is a sparse image   
INFO: Looking for file 'system.img'
......
system.img is a sparse image 
{<program> (68.80 MB) 140912 sectors needed at location 3031040 on LUN 0} 
{<program> (8.00 KB) 16 sectors needed at location 3293184 on LUN 0} 
{<program> (8.00 KB) 16 sectors needed at location 3293320 on LUN 0} 
{<program> (119.97 MB) 245696 sectors needed at location 3297336 on LUN 0

SlarkXiao, 2024-08-16:
Actually this is issue was caused by the sparse image format. We shall transfer it to RAW image format.

andersson, 2025-09-09:
Sparse image support has been implemented since the time this issue was opened. Please file new issue if problem remains.