linux-msm/qdl #78: USB: using out-chunk-size of 1048576

sifat06, 2024-07-29:
What does this mean? When I enter 
`sudo "/home/sifat06/Downloads/LENOVO yt3-850m/qdl/qdl" --debug "/home/sifat06/Downloads/LENOVO yt3-850m/YT3_850M_S000139_220803_ROW/prog_emmc_firehose_8909_ddr.mbn" "/home/sifat06/Downloads/LENOVO yt3-850m/YT3_850M_S000139_220803_ROW/rawprogram_unsparse.xml" "/home/sifat06/Downloads/LENOVO yt3-850m/YT3_850M_S000139_220803_ROW/patch0.xml"` 
It shows:
`USB: using out-chunk-size of 1048576`
and the command finishes...

sifat06, 2024-07-30:
No idea?

jackestar, 2024-09-03:
In my case in archlinux both by AUR package and compiling the code from this repository, with a Redmi 9T (lime)

I had the same "problem", I don't know if the program is designed to work under some "programmer", it is assumed that under USB the maximum chunk size is 512 bytes or 1024 in some cases, the same with payload (im not 100% sure) . BTW, in the file 

https://github.com/linux-msm/qdl/blob/cbd46184d33af597664e08aff2b9181ae2f87aa6/firehose.c#L201

Here the "maximum" is declared, in my case I changed it to 512 and it doesn't show me the error message. **BUT** it still doesn't work, in my case at first it showed the message `usb: using out-chunk-size of 1048576`, I tried to perform the actions and it never finished them, after the corrections, it showed me the same output in the console but without `usb: using out-chunk-size of 1048576` (and without working).

Eventually I found another tool that did work for me.

[bkerler/edl](https://github.com/bkerler/edl)

It may be a matter of use case or the specific device, IDK

closed 2024-12-12 by andersson