linux-msm/qdl #83: qdl: failed to initialize libusb: Permission denied

rarenicks, 2024-11-21:
Hi,
I am trying to recover my Qualcomm RB5 device using QDL model
I am able to see the device in lsusb : 
Bus 001 Device 004: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)

Also set the proper udev rules for the device, still facing the permission issue. 
"qdl: failed to initialize libusb: Permission denied"

I have given all required permission for the device, but the issue looks not with the device permission but with the qdl not being able to use libusb. 

Host system : 
Description:	Ubuntu 22.04.5 LTS
Linux helios 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

Installed qdl using snap "snap install qdl"

Thanks,
Avdhesh

rarenicks, 2024-11-21:
Hi 
I was able to run qdl after building it from github source 

During flashing "./qdl prog_firehose_ddr.elf rawprogram?.xml patch?.xml" 

Got this error : 

[PROGRAM] flashed "aop_a" successfully
LOG: INFO: Calling handler for program
[PROGRAM] flashed "tz_a" successfully
LOG: INFO: Calling handler for program
[PROGRAM] flashed "hyp_a" successfully
**Unable to open BTFM.bin...failing**

Do you have any idea about this error or point me in the right direction ?

ndechesne, 2024-11-21:
> Hi I was able to run qdl after building it from github source
>

Are you saying that you have a different behavior when you build from source vs when you are using the Ubuntu snap package? Or have you also modified the udev rules, as I pointed out earlier? 

 
> During flashing "./qdl prog_firehose_ddr.elf rawprogram?.xml patch?.xml"
>
> Got this error :
>
> [PROGRAM] flashed "aop_a" successfully LOG: INFO: Calling handler for program [PROGRAM] flashed "tz_a" [...]

the BTFM.bin file is referenced in one the rawprogram XML files, and needs to be 'flashed', but is missing from your local folder.

andersson, 2024-12-12:
> Hi I was able to run qdl after building it from github source
> [..]
> **Unable to open BTFM.bin...failing**
> 
> Do you have any idea about this error or point me in the right direction ?

QDL used to simply skip flashing partitions where it could find the files referenced, with the side effect of users reporting that what seemed to be a successful flash would actually have left old content in some of the partitions. So the tool will now instead abort if it can not find any of the referenced files.

You can opt in to the old behavior by specifying the **--allow-missing** argument when launching qdl.

Looks like your original issue has been resolved as well, so I'm closing this for now.

Thanks,
Bjorn

closed 2024-12-12 by andersson