linux-msm/qdl #97: Implement support for flashing sparse images

obbardc, 2025-03-22:
There are a number of open PRs trying to implement sparse image format:

- https://github.com/linux-msm/qdl/pull/37
- https://github.com/linux-msm/qdl/pull/30
- https://github.com/linux-msm/qdl/pull/6
- https://github.com/linux-msm/qdl/pull/5

It'd be really nice to get _something_ merged which implements this and close out the other PRs :-)

obbardc, 2025-03-22:
cc @lumag @ndechesne @andersson ; not sure if any of you are interested in this feature?

andersson, 2025-03-26:
@obbardc it does sound like a good feature, and the number of PRs just confirms that there is user demand.

I didn't like the initial attempts due to their added dependencies.
I think #37 looks good implementation wise, and I wanted to dig into this further, but I don't have any sparse use cases and my attempt at trying it out didn't give me the expected results (perhaps due to user error, perhaps because we don't have erase-before-flash support?), and I didn't invest the necessary time to figure out why.

I'd be happy to see #37 move forward.

kismeter, 2025-06-15:
using #37 , the device boot has no issues, but adb remount will cause device not boot

andersson, 2025-06-30:
#37 has been merged. Please retest and reopen (or open a new ticket) if it doesn't work as expected.

closed 2025-06-30 by andersson