linux-msm/qdl #107: Cannot compile due to -Wimplicit-function-declaration

KarthikJay, 2025-05-16:
usb.c: In function ‘usb_init’:
usb.c:295:34: error: implicit declaration of function ‘malloc’ [-Wimplicit-function-declaration]
295 |         struct qdl_device *qdl = malloc(sizeof(struct qdl_device_usb));
|                                  ^~~~~~
usb.c:11:1: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’
10 | #include "qdl.h"
+++ |+#include <stdlib.h>
11 |
usb.c:295:34: warning: incompatible implicit declaration of built-in function ‘malloc’ [-Wbuiltin-declaration-mismatch]
295 |         struct qdl_device *qdl = malloc(sizeof(struct qdl_device_usb));
|                                  ^~~~~~
usb.c:295:34: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’

Tried installing via AUR qdl-git script.
https://aur.archlinux.org/packages/qdl-git

ndechesne, 2025-05-16:
#106 should fix this issue. Hopefully should be merged soon.

andersson, 2025-06-23:
#106 has been merged, @KarthikJay can you please confirm that it's okay to close this issue?

KarthikJay, 2025-06-23:
Confirmed that it fixed this issue!

Thanks for the response!